ch32-rs/wlink #5: Faild to found entry on Windows OS

tonyho, 2023-03-11:
Using in Windows 10 LTSC with Administrator permission
Error Log:

```
.\wlink.exe
Error: USB error: Entity not found

Caused by:
    Entity not found
```

I can use MountRiver IDE to burn in the same environment even if the wlink command line failed, so the USB and the wch link tool are OK.

UCSIG, 2023-04-13:
You have to change the driver that Windows uses to access the WCH-Link to WinUSB (or libusb-win32), so that the wlink tool can talk to the WCH-Link device. 
You can do that with the Zadig tool. In Zadig, list all devices by selecting `Options -> List All Devices`. Now you should be able to see `WCH-Link (Interface 0)` and` WCH-Link (Interface 1)` in the devices dropdown list. 
Select `WCH-Link (Interface 0)`. The currently installed driver will most likely be WCHLink_A64. Select WinUSB as the new driver and click Replace Driver to replace the currently installed one with WinUSB. Now you should be able to use the wlink cli tool. 
However, now you can't use MounRiver Studio anymore to flash the mcu. In order to be able to use MounRiver again, use their driver installer to replace the WinUSB driver with their driver. The installer can be found in the programs installation directory under LinkDrv\WCHLink\DRVSETUP64\DRVSETUP64.exe.

andelf, 2023-10-03:
I should have added a better error description.

andelf, 2023-11-05:
I'm writing an interface for the official Windows driver. The only drawback is.... it should be 32-bit.

andelf, 2024-01-07:
Could you help to check the nightly release?
I've added a windows native driver support.

Plecra, 2024-01-16:
I've tried the new windows driver - plenty of features work :) But serial debug print doesn't seem to get properly forwarded to the COM port

andelf, 2024-01-17:
@Plecra , Thanks for your feedback.

I'm looking for a method to simplify command lines.
An acceptable approach would be to include commonly used combinations in the `--help` doc.

andelf, 2026-11-12:
Same class of problem as #57: the probe was bound to a driver nusb could not open.

wlink now supports WCH's native Windows driver (`WCHLinkDrv`, merged in #116) alongside the legacy CH375 DLL and WinUSB, so the probe can be used without installing/replacing a WinUSB driver. See the README section "Windows drivers" (and the new troubleshooting entry from #137).

Please retry with the next nightly/0.1.4 build, and reopen with `wlink -vv list` output if it still fails.

closed 2026-11-12 by andelf