ch32-rs/wlink #8: Flash address

Schildkroet, 2023-05-18:
Hello,
would it be possible to add an address option to the flash command?

e.g. wlink flash 0x00000200 app.bin

closed 2023-05-20 by Schildkroet

Schildkroet, 2023-05-20:
How do i now pass flash address?
This does not work anymore:
`flash 0x08000000 ./firmware.bin`

andelf, 2023-05-21:
> Can you please update the Readme. There it is still wrong:

Got it.

DBRJ-A88, 2023-05-25:
When I run the revised address command it works like a charm but the "--help" still does not add "-a" to the options list?

wlink flash -a 0x08000000 ./CH32V203C8T6-Blinking.hex
03:54:06 [INFO] WCH-Link v2.8 (WCH-LinkE-CH32V305)
03:54:06 [INFO] Attached chip: CH32V20x(0x20310510)
03:54:06 [INFO] Read ./CH32V203C8T6-Blinking.hex as IntelHex format
03:54:06 [INFO] Flashing 7212 bytes to 0x08000000
03:54:07 [INFO] Flash done
03:54:07 [INFO] Now reset...
03:54:08 [INFO] Resume executing...

Suggested Revised "wlink Help" message follows:

Invocation: db@wks-32:~/Desktop$ wlink help

WCH-Link flash tool for CH32V faimily MCUs

Usage: wlink [OPTIONS] [COMMAND]

Commands:
  dump       Dump memory region
  regs       Dump registers
  erase      Erase flash
  flash      Program the flash
  write-reg  Force set register
  write-mem  Force write a memory word
  halt       Halts the MCU
  resume     Resumes the MCU
  reset      Reset the MCU
  status     Debug, check status
  help       Print this message or the help of the given subcommand(s)

Options:
  -d, --device <INDEX>  Optional device index to operate on
  -v, --verbose...      Turn debugging information on
      --detach          Detach chip after operation
  -h, --help            Print help
  -V, --version         Print version
  -a                    Load Address Example: wlink flash -a 0x08000000 ./file.hex    ****************************<<<

Schildkroet, 2023-05-25:
You have to specify the command:
`wlink flash --help`