ch32-rs/wlink #12: Update the command line  "help" listing, please! :)

DBRJ-A88, 2023-05-25:
When I run the revised address command it works like a charm but the "--help" still does not add "-a" to the options list?

wlink flash -a 0x08000000 ./CH32V203C8T6-Blinking.hex
03:54:06 [INFO] WCH-Link v2.8 (WCH-LinkE-CH32V305)
03:54:06 [INFO] Attached chip: CH32V20x(0x20310510)
03:54:06 [INFO] Read ./CH32V203C8T6-Blinking.hex as IntelHex format
03:54:06 [INFO] Flashing 7212 bytes to 0x08000000
03:54:07 [INFO] Flash done
03:54:07 [INFO] Now reset...
03:54:08 [INFO] Resume executing...

Suggested Revised "wlink Help" message follows:

Invocation: db@wks-32:~/Desktop$ wlink help

WCH-Link flash tool for CH32V faimily MCUs

Usage: wlink [OPTIONS] [COMMAND]

Commands:
dump Dump memory region
regs Dump registers
erase Erase flash
flash Program the flash
write-reg Force set register
write-mem Force write a memory word
halt Halts the MCU
resume Resumes the MCU
reset Reset the MCU
status Debug, check status
help Print this message or the help of the given subcommand(s)

Options:
-d, --device Optional device index to operate on
-v, --verbose... Turn debugging information on
--detach Detach chip after operation
-h, --help Print help
-V, --version Print version
-a Load Address Example: wlink flash -a 0x08000000 ./file.hex ****************************<<<

KurtE, 2023-05-25:
Sorry I am a novice using this tool, but when I tried it yesterday, I found:
```
C:\Users\kurte>wlink flash --help
Program the flash

Usage: wlink flash [OPTIONS] <PATH>

Arguments:
  <PATH>  Path to the binary file to flash

Options:
  -a, --address <ADDRESS>  Address in u32 [default: 134217728]
      --detach             Detach chip after operation
  -h, --help               Print help

C:\Users\kurte>
```

closed 2023-05-25 by septs

andelf, 2023-05-26:
This tool uses `clap` to handle commands. So it's a multi-level subcommand style.