ch32-rs/wlink #13: missing Cargo.toml error

gicking, 2023-05-28:
Hello,

I would like use WLINK to upload code to CH32Vxxx under Linux and verify it. I installed it using
```
sudo apt install cargo
cargo install --git https://github.com/ch32-rs/wlink
```
as described on the Readme.md.

But when I execute 
```
cargo run -- flash --address 0x08000000 ./firmware.bin`
```

I just get an error message: 
```
error: could not find 'Cargo.toml' in 'PRJDIR' or any parent directory
```
with PRJDIR the CWD.

I searched the internet and tried [this answer](https://stackoverflow.com/a/74167679), which installed a lot of rust libraries
```
cargo install --list | 
awk '/^\w/ { print $1 }' | 
while read x; do cargo install "$x"; done
```
but without success. What am I missing?

My configuration:
- Xubuntu 22.04.2 LTS
- cargo 1.65.0
- rustc --version

Thanks a lot in advance!

andelf, 2023-05-28:
Sorry for the misleading.
After a `cargo install`, you can run it with `wlink flash .....`

gicking, 2023-05-28:
thanks for the fast response! 

However, when I try `wlink flash --address 0x1FFFF000 firmware.bin`, I get an error message that command `wlink` cannot be found.  

And `cargo run wlink...` just yields the same error message as before.

Sorry, but I have never used cargo or rust before! So please do not assume any prior knowledge...    ;-)

gicking, 2023-05-29:
ok, I made some progress  :-)   I found `wlink` under `~/.cargo/bin/wlink`. But when I try to flash a bin file I get this error message
```
~/.cargo/bin/wlink flash firmware.bin
Error: Unknown WCH-Link variant: 18
```

Here's the output of `lsusb -v` for my WCH-Link, maybe that helps:
```
Bus 001 Device 003: ID 1a86:8010 QinHeng Electronics WCH-Link
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x8010 
  bcdDevice            2.08
  iManufacturer           1 wch.cn
  iProduct                2 WCH-Link
  iSerial                 3 66448F06EB21
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0078
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         1
      bFunctionClass        255 Vendor Specific Class
      bFunctionSubClass       0 
      bFunctionProtocol       0 
      iFunction               2 WCH-Link
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           4
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    128 
      bInterfaceProtocol     85 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         1
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               4 WCH-Link
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        1
        bSlaveInterface         2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

gicking, 2023-05-29:
Now it worked. After replacing `~/.cargo/bin/wlink` with the nightly build from https://github.com/ch32-rs/wlink/releases/tag/nightly, it worked right away.

Thanks a lot!

PS: maybe you should update the repo Readme.md for dummies like me...!? Saves you some work

closed 2023-05-29 by gicking