ch32-rs/wlink #14: Enable/disable Read-Out protection?

gicking, 2023-06-03:
Hello,

is it possible to enable and disable the CH32Vxx flash read-out protection (see image below)? I understand that - at least for CH32V003 - protection is enabled by default, so I assume this must be possible somehow...?

[image]

For your advice thanks a lot in advance!

Greetings from Germany, Georg

andelf, 2023-06-04:
It requires some reverse engineering. I'll check this.

septs, 2023-06-09:
OpenOCD source code: https://github.com/xu7wong/openocd_wchlink

from https://www.wch.cn/bbs/thread-91946-1.html

archive: [openocd-wchlink.bundle.zip](https://github.com/ch32-rs/wlink/files/11699165/openocd-wchlink.bundle.zip)

andelf, 2023-07-02:
@gicking @septs 
After digging into the WCH-LinkUtility UI, Flash protect/unprotect is only supported by CH32V003/V103/V20x/V30x/X035.

This is implemented in the newest main branch. You can check if it works.

I haven't got a CH32X035. 😢 So CH32X035 is not supported yet.

gicking, 2023-08-01:
hi,

sorry for the long delay! I bricked my CH32V003 and had to wait for a replacement chip. 

I just tried protect/unprotect on the (new) CH32V003 and I can confirm that it works in my setup (Linux Ubuntu 22.04 with WCH-LINKE-R0-1v3. Thanks a lot!

Currently I have no other device to test it with, so should I close this issue...?

andelf, 2023-08-02:
@gicking Thanks for I'll close this. I tested this on my CH32V003-EVT.

closed 2023-08-02 by andelf

smaharj1, 2024-01-13:
@andelf is CH32X035 supported yet? I tried to unprotect for X035, and it did not do anything.

andelf, 2024-01-14:
@smaharj1  unprotect & protect changes since old version of wlink.
So I spent some time digging into the new version.