ch32-rs/wlink #15: Is there a way to unlock the flash of the different boards.

KurtE, 2023-06-03:
That is some of the example boards I have received have their flash locked.  I needed to use the WCH Link utility
[image]
unlock button to allow me to program the boards.

But after I update the USB driver to work with wlink, that become more problematic... So wondering if there is a command line way to do this.

Thanks
Kurt

andelf, 2023-06-04:
I unlock the flash using https://github.com/ch32-rs/wchisp
It has a predefined config register map.

KurtE, 2023-06-04:
Thanks, I will have to try it out. 

Looks like all of the artifacts up there have expired.  So will see what it takes to try to build it.

closed 2023-06-11 by KurtE