ch32-rs/wlink #16: Error: DM Abstract comand error: Parity on CH32V307V-R1-1v0 board (CH32V307 VCT6)

lennyerik, 2023-06-05:
I have a CH32V307V-R1-1v0 devboard (second revision of the CH32V307 evaluation board) running WCH-Link 2.9 and am trying to flash it using wlink.

All operations supported by wlink (flashing, reading and writing registers, resetting, ...) seem to work fine on the first run after power-cycling the board. However, subsequent runs of the same command (especially when executing the same command multiple times in quick succession) sometimes result in the error `Error: DM Abstract comand error: Parity`.

This issue occurs with all subcommands and I managed to track it down to [this check in operations.rs](https://github.com/ch32-rs/wlink/blob/8f5ab27e1bb9306b75aad7e8df35f1d01d9402be/src/operations.rs#L371). As I am not very familiar with the WCH-Link protocol, I'm not sure on what exactly could be causing this problem, although it looks to me like `read_reg` is called on chip initialisation somewhere and fails sometimes because `ensure_mcu_halt` caused an abstractcs cmd error somehow.

I added a [temporary fix](https://github.com/lennyerik/wlink/commit/4857702a84a960d87213d9bc0a12b6cff155e9a3) in my own fork of the repo which just clears the abstractcs cmderror queue on each call to `ensure_mcu_halt`. This seems to have fixed the issue, but I'm hesitant to open a PR, since I'm unsure whether this is actually a fix for the root cause of the issue and not just a workaround. I'd appreciate it if someone more familiar with WCH-Link could have a look at the fix first.

Full log of  `wlink -v regs` from when this issue occurs:
```
16:56:10 [INFO] WCH-Link v2.9 (WCH-Link-CH549)
16:56:10 [INFO] Attached chip: CH32V30x(0x30700518)
16:56:10 [DEBUG] (1) wlink::commands: TODO: cannot read chip id
16:56:10 [DEBUG] (1) wlink::operations: Chip UID: 00-00-00-00-00-00-00-00
16:56:10 [DEBUG] (1) wlink::operations: Flash protected: false
16:56:10 [DEBUG] (1) wlink::operations: SRAM CODE mode: 2
Error: DM Abstract comand error: Parity
```
Note that the `cannot read chip id` log is most likely unrelated to this issue as it happens with this board regardless of whether the `Abstract command error` occurs.

Schildkroet, 2023-06-12:
I also ran into this problem. For me it happens when i send alot data through serialport on debugger. Unplugging the debugger clears the issue.

andelf, 2023-06-13:
Thanks for the report. I'll check this on my CH32V307 board later.

andelf, 2023-06-27:
Resolved. 
"QingKeV2 Microprocessor Debug Manual" didn't mention a step used in openocd code to clear cmderr state.

andelf, 2023-06-27:
@lennyerik  Could you try the fixed version?
It should be ready when nightly build released.

lennyerik, 2023-06-30:
The fix also works on my board. :+1: 
Thanks for taking the time to look into this issue!

closed 2023-06-30 by lennyerik