ch32-rs/wlink #17: Program the processor without erasing whole chip?

KurtE, 2023-06-11:
Can wlink program a chip like the CH32V308, without doing a full chip erasure? 
And if not, could it be done?

Why, if the chip lets say has 256KB of flash and my program maybe use 64KB, In some case I might want to use some of the remaining flash as a file system,  probably using LittleFS.   Likewise I might want to reserve some portion of flash to emulate an EEPROM.

Possible?

Thanks
Kurt

Schildkroet, 2023-06-12:
AFAIK wlink only programms the given address+length

KurtE, 2023-06-13:
Thanks, I will double check.  
...
Updated - it appears like you are right.  It looks like the region of memory was not touched.  

Now mostly using  OpenOCD WCH version to update the chip and that appears to wipe the whole chip.  Not sure if something in the script controls it or not... But that is a different project to look at.

Thanks again

closed 2023-06-13 by KurtE