ch32-rs/wlink #20: some more protocol decoding

usbman01, 2023-07-05:
I am working on ARM version of ulink. It seems when using the WCH-Link utility the same protocol is used
So here is the 0x0c cmd. Maybe this can help on the RISC part too.
request:
              cmd       len         4  | 8             0 .. 3
<0x81> <0x0C> 0x02 <Target CPU> <speed>

response:
               cmd        len
<0x82> <0x0C> <0x01> <OK>

Target CPU 0x04 = F103
                   0x08 = F203

Speed 1..3 means slow medium fast
Speed 0 maybe default? 
OK response = 0x01

andelf, 2023-07-06:
Many thanks:
I'll add the docs.

https://github.com/ch32-rs/wlink/blob/085b39b8015c5cbbda8cfe197478ab3217e1666d/src/operations.rs#L32-L36

andelf, 2023-07-06:
I need a better scheme to document the protocol. @septs Any ideas?

andelf, 2023-07-22:
close as completed

closed 2023-07-22 by andelf