ch32-rs/wlink #24: Error: DMI Status Failed

chihiro1234567, 2023-08-05:
When flashing to the chip with wlink, 'DMI Status Failed' is displayed. (Sometimes it does not show up)

|||
|--|--|
|Writer|WCH-LinkE|
|chip|CH32V203K8T6|

1. get sample rust-ch32 program

```
> git clone https://github.com/ch32-rs/ch32v203-demo.git
> cd ch32v203-demo
> rustup target add riscv32imac-unknown-none-elf
```

2. Edit Cargo.toml, File size reduction is the goal.

```toml
[profile.release]
codegen-units = 1
debug = false
debug-assertions = false
overflow-checks = false
panic = 'abort'
lto = true
opt-level = "z"
incremental = false
strip = true
```

3. Build program

```
> cargo build --release
```

4. Get 'riscv64-unknown-elf-objcopy', and create Firmware

```bash
> sudo apt install binutils-riscv64-unknown-elf
> cd target/riscv32imac-unknown-none-elf/release
> riscv64-unknown-elf-objcopy -O binary ch32v203-demo firmware.bin
```

5. Get wlink, and flash firmware.

```
> cargo install --git https://github.com/ch32-rs/wlink
> wlink
16:22:52 [INFO] Found WCH-LinkRV, Bus 002 Device 008: ID 1a86:8010
No command given, use --help for help.

> wlink flash --chip CH32V20X --address 0x08000000 ./firmware.bin
00:13:11 [INFO] WCH-Link v2.9 (WCH-LinkE-CH32V305)
00:13:11 [INFO] Attached chip: CH32V20X(0x20320500)
00:13:11 [INFO] Read ./firmware.bin as Binary format
00:13:11 [INFO] Flashing 624 bytes to 0x08000000
00:13:11 [INFO] Flash done
00:13:11 [INFO] Now reset...
00:13:12 [INFO] Resume executing...
Error: DMI Status Failed
```
After the reset instruction, waiting for a return and no response? I guess so.
If so, is there a problem with the circuit?
I am having trouble finding an example that does not use a development board.

The schematic is as follows.

* capacitor 0.1uF ( VDD-GND)
* resistor 10kΩ ( VDD-NRST )
* VDD, VDDA => 3.3V
* GND => GND

[image]

andelf, 2023-08-05:
Thanks for your report.
This is caused by missing reattach after a successful flash.

andelf, 2023-08-05:
Could you check if it fixed?

I've tested on my nanoCH32V203(CH32V203C8T6) and FlappyBoard(CH32V203G6U6) for the blinky example.

chihiro1234567, 2023-08-06:
Thanks for fix.
I tried again.

```bash
> cargo install --git https://github.com/ch32-rs/wlink

> wlink flash --chip CH32V20X --address 0x08000000 ./firmware.bin

08:22:21 [INFO] WCH-Link v2.9 (WCH-LinkE-CH32V305)
08:22:21 [INFO] Attached chip: CH32V20X(0x20320500)
08:22:21 [INFO] Chip UID: cd-ab-6d-76-55-bc-8f-de
08:22:21 [INFO] Flash protected: false
08:22:21 [INFO] RISC-V ISA: Some("RV32ACIMUX")
08:22:21 [INFO] RISC-V arch: Some("WCH-V4B")
08:22:21 [INFO] Read ./firmware.bin as Binary format
08:22:21 [INFO] Flashing 624 bytes to 0x08000000
08:22:21 [INFO] Flash done
08:22:22 [INFO] Now reset...
08:22:22 [WARN] Chip already attached
08:22:22 [INFO] Attached chip: CH32V20X(0x20320500)
08:22:22 [INFO] Resume executing...
```

The error no longer occurs.
I think that's good.

closed 2023-08-06 by chihiro1234567