ch32-rs/wlink #26: TODO: Clear All Code Flash by NRST

comato, 2023-09-09:
Is it possible to implement Clear All Code Flash-By Pin NRST and Clear All Code Flash-By Power Off?
I using Platformio under Linux. If CH32V003 is in sleep mode, it is not possible to upload firmware. 
One of solution is add about 5sec delay to start of program (before sleep command), and upload firmware immediately after pushing reset button (if is available on board). But it is not ideal solution.
Or plug WCH LinkE to Windows PC with running WCH-Link utility, and use Clear All Code Flash-By Pin NRST or Clear All Code Flash-By Power Off... but it is terrible.
Thanks
[image]

andelf, 2023-10-03:
This is implemented. I've tested it using WCH-LinkE and the official CH32V003F4P6-R0 board.

ohdarling, 2023-10-13:
> This is implemented. I've tested it using WCH-LinkE and the official CH32V003F4P6-R0 board.

Is there any example for using "Clear All Code Flash-By Power off" with wlink? Thanks.

ohdarling, 2023-10-13:
I found the command parameters:

`wlink erase --method power-off`

However, it doesn't seem to work, and it still prompts an error:

```
wlink -v erase --method power-off
17:19:28 [DEBUG] (1) wlink::device: send command: GetProbeInfo
17:19:28 [INFO] WCH-Link v2.10 (WCH-LinkE-CH32V305)
17:19:28 [DEBUG] (1) wlink::device: send command: SetSpeed { riscvchip: 1, speed: High }
17:19:28 [DEBUG] (1) wlink::device: send command: AttachChip
17:19:28 [DEBUG] (1) wlink::operations: retrying...
17:19:28 [DEBUG] (1) wlink::device: send command: SetSpeed { riscvchip: 1, speed: High }
17:19:28 [DEBUG] (1) wlink::device: send command: AttachChip
17:19:28 [DEBUG] (1) wlink::operations: retrying...
17:19:28 [DEBUG] (1) wlink::device: send command: SetSpeed { riscvchip: 1, speed: High }
17:19:28 [DEBUG] (1) wlink::device: send command: AttachChip
17:19:28 [DEBUG] (1) wlink::operations: retrying...
Error: Probe is not attached to an MCU, or debug is not enabled. (hint: use wchisp to enable debug)
```

It seems that in the code, there are "set_speed" and "attach_chip" operations executed before performing the "erase" operation. Therefore, the power-off command did not take effect.

https://github.com/ch32-rs/wlink/blob/43e89353be96a61d389e8bf531bb91923a911c9e/src/main.rs#L170-L174

Thank you.

andelf, 2023-10-17:
Close as implemented

commit 0eb7b4ce (2023-10-03): feat: erase by RST pin

Close #26
closed 2023-10-17 by andelf