ch32-rs/wlink #27: Unsupported wchlink

9names, 2023-10-01:
Hardware is WCH-LinkE-R0-1v3. Bought from Aliexpress, from WCH Official Store as part of CH32V003/203-LinkE Kit.
Tried with stock firmware (2.9), also updated to latest firmware 2.10 with MounRiver Studio.

wlink reports:
Error: Unknown WCH-Link variant: 18

Tested using Ubuntu 23.04

9names, 2023-10-01:
lsusb output (in wchlink mode)
```
$ lsusb -d 1a86:8010 -v

Bus 003 Device 029: ID 1a86:8010 QinHeng Electronics WCH-Link
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x8010 
  bcdDevice            2.10
  iManufacturer           1 wch.cn
  iProduct                2 WCH-Link
  iSerial                 3 00A08F068573
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0078
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         1
      bFunctionClass        255 Vendor Specific Class
      bFunctionSubClass       0 
      bFunctionProtocol       0 
      iFunction               2 WCH-Link
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           4
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    128 
      bInterfaceProtocol     85 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         1
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               4 WCH-Link
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        1
        bSlaveInterface         2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

9names, 2023-10-01:
Output from probe-rs wlink fork
I added an extra print line for more version info:
```rust
tracing::warn!("WCH-Link info: {} {} {}", self.v_major, self.v_minor, rxbuf[5]);
```
```
DEBUG open: probe_rs::probe::wlink: Initializing WCH-Link...
DEBUG open: probe_rs::probe::wlink: Getting version of WCH-Link...
TRACE open: probe_rs::probe::wlink::usb_interface: Sending command [81, 0d, 01, 01] to WCH-Link, timeout: 300ms
TRACE open: probe_rs::probe::wlink::usb_interface: Receive response [82, 0d, 04, 02, 0a, 12, 00] from WCH-Link
 WARN open: probe_rs::probe::wlink: WCH-Link info: 2 10 18
 INFO probe_rs: Wrote log to "/home/nine/.local/share/probe-rs/1696151787118.log"
Error: Failed to open the debug probe.

Caused by:
    0: An error specific to a probe type occurred
    1: Unknown WCH-Link device(new variant?)
```

9names, 2023-10-02:
I downgraded the firmware to 2.7 from https://github.com/NgoHungCuong/WCH-LinkE-Updater/tree/main/Firmware
Now it's not throwing errors any more:
```
$ wlink
00:13:24 [INFO] WCH-Link v2.7 (WCH-LinkE-CH32V305)
No command given, use --help for help.
```

So I guess it's just incompatible with firmware versions >= 2.8?

andelf, 2023-10-02:
I'm working on a refactor.
I'll update the supported firmware version to v30 soon(actually it's a 0x020a, WCH calculates it as `buf[0] * 10 + buf[1]`).

andelf, 2023-10-02:
I'm planning only to support the newest probe in the future. The firmware is forced to be updated to the most recent by WCH's toolchain.

For a debug or tracing session, use `wlink -vv`.

andelf, 2023-10-03:
@9names  Could you help test the new version?(not released yet, avaliable via the main branch)

PR #28 (2023-10-03): New chip and probe support
Close #25
Fix #27