ch32-rs/wlink #29: TODO: CH32L103 support

andelf, 2023-10-03:
I think this is implemented in #28.
I haven't got a chip or EVT board to verify this.

21km43, 2024-09-28:
I think we can close this issue

cjacker, 2024-11-28:
Confirm works with:
WCH Official ch32vl103c8t6 evt board.
WCH-LinkE r0 1v3 and r0 1v2 with latest firmware.

andelf, 2024-11-28:
@cjacker Thanks for your confirmation.

closed 2024-11-28 by andelf

cjacker, 2024-11-28:
@andelf  Thanks for your great work.

And the major feature missing is #42 "Upgrade WCH-Link/E firmware online", hope it comes soon :-).

Now I have to use another WCH-Link/E to upgrade firmware under linux.

[Ghidra](https://github.com/NationalSecurityAgency/ghidra) may help to analyze the protocol. Recently I use Ghidra to analyze 'clearcodeflash' related functions of WCH-LinkUtility/McuCompilerDll.dl and try to implement 'erase by NRST/by PowerOff' to rescue a bricked board, After done, I found wlink already done:-D