ch32-rs/wlink #41: SDI not working with  WCH-Link v2.11 (WCH-LinkE-CH32V305)

thirstyone, 2023-11-23:
here are the last lines: 

```
14:30:19 [INFO] Now connect to the WCH-Link serial port to read SDI print
er computer software

```
(found these lines in the end of WCH-LinkE-APP-IAP.bin)

andelf, 2023-11-23:
I've seen this error while developing the feature.
If data0 is not configured properly. 
The UART link will print(It's almost always corrupted text):

> Please check the SDl, and Enable this function through the upp**er computer software**

- A common pitfall is misuse of little endian and big endian while writing data0.
- And meanwhile, data0 should be written in 32-bit word.
- Write data1, then data0

thirstyone, 2023-11-23:
Could you please provide a working example of SDI print? After a while I get a bricked mcu (the second time!), so that I have to revert to the original wch link utility to " clear all code flash - by power off". (BTW, is it possible to add such a feature to this wlink?)

thirstyone, 2023-11-23:
btw, have you checked this feature with v2.11?

andelf, 2023-11-24:
Checkout: https://github.com/ch32-rs/ch32v00x-hal/blob/main/src/debug.rs
And: https://github.com/ch32-rs/ch32v00x-hal/blob/main/examples/debug.rs

```console
> wlink erase --help
Erase flash

Usage: wlink erase [OPTIONS]

Options:
      --method <METHOD>
          Erase mode

          [default: default]

          Possible values:
          - power-off: Erase code flash by power off, the probe will power off the target chip
          - pin-rst:   Erase code flash by RST pin, the probe will active the nRST line. Requires a RST pin connection
          - default:   Erase code flash by probe command
....
```

kaidegit, 2023-11-24:
> Could you please provide a working example of SDI print? After a while I get a bricked mcu (the second time!), so that I have to revert to the original wch link utility to " clear all code flash - by power off". (BTW, is it possible to add such a feature to this wlink?)

https://www.wch.cn/bbs/thread-107881-1.html

thirstyone, 2023-11-24:
ha! I should have checked that earlier (i mean erase --help)
However: 
`CH32V003-makefile-example-main % wlink erase --method power-off --chip ch32v003    
07:07:01 [INFO] WCH-Link v2.11 (WCH-LinkE-CH32V305)
07:07:01 [INFO] Erase chip by PowerOff
07:07:01 [ERROR] --chip not specified
Error: Chip doesn't support power off erase
`
it seems though, --chip option is working, since when I use a different chip name, it produces another error: 
` wlink erase --method power-off --chip CH32V003J4M6
error: invalid value 'CH32V003J4M6' for '--chip <CHIP>': Unknown RISC-V Chip: 0x00
`

thirstyone, 2023-11-24:
basically, the following code in main.c should produce some results, shouldn't it? 
`
#define DEBUG_DATA0_ADDRESS  ((volatile uint32_t*)0xE00000F4)
#define DEBUG_DATA1_ADDRESS  ((volatile uint32_t*)0xE00000F8)

int main(void)
{
    NVIC_PriorityGroupConfig(NVIC_PriorityGroup_2);
	    Delay_Init();
		Delay_Ms(4000);		
		USART_Printf_Init(115200);
		*(DEBUG_DATA0_ADDRESS) = 0;
		Delay_Ms(1000);
		*(DEBUG_DATA1_ADDRESS) = 0x55565758;
		*(DEBUG_DATA0_ADDRESS) = 0x07505151;
//		*(DEBUG_DATA0_ADDRESS) = 0x50505107; //tried this as well...
		
	while(1) {
		int i = 0;
			Delay_Ms(1000);
	}
```
but I get the same 'er computer software' message (sometimes different part of it, but anyway, not what I expect)

thirstyone, 2023-11-30:
I have no idea what was the culprit. Probably, some segfault because of misuse of printf library. But now it works as expected.  Now people ask if it's possible to use the scan functionality with SDI ;-)  You know, the more you have, the more you want.