ch32-rs/wlink #42: Feature Request: Upgrade WCH-Link firmware

andelf, 2023-11-23:
esp. for macOS users.

sass20, 2024-02-18:
hi
Can you add show virsion of firmware in wchlink?

andelf, 2024-02-19:
It's already displayed in basic logging:

> 16:26:11 [INFO] Connected to WCH-Link **v2.11(v31)** (WCH-LinkE-CH32V305)

v31 is used by official tools.
v2.11 is the original version. => `0x020a`

sass20, 2024-02-19:
I connect wch-linkE only, wutout targer cpu and run wlink.exe -v

wlink.exe -v
19:51:13 [INFO] Found WCH-LinkRV, Bus 001 Device 005: ID 1a86:8010
No command given, use --help for help.
hint: use `wlink status` to get started.

What is options need place for show  [INFO] Connected to 

thank you

andelf, 2024-02-20:
A new "list" subcommand is added recently.

orangecms, 2024-11-03:
I have the first WCH-Link, the one based on CH549.
I want to flash a CH32V003, which is currently not supported.

So I tried upgrading the WCH-Link firmware, using the file `WCH-LinkUtility/Firmware_Link/FIRMWARE_CH549.bin` from the WCH Link Utility zip file with the J1 switch and `wchisp`. Now it is no longer showing up on USB. There are also the following files:
```
FIRMWARE_CH32V203.bin  FIRMWARE_CH549.bin      WCH-DAPLink_APP_IAP.bin   WCH-LinkE-APP-IAP.bin
FIRMWARE_CH32V208.bin  FIRMWARE_DAP_CH549.bin  WCH-Link_APP_IAP_ARM.bin  WCH-LinkW-APP-IAP.bin
FIRMWARE_CH32V305.bin  firmware_version.txt    WCH-Link_APP_IAP_RV.bin   wchlink.wcfg
```

Do you have any idea? Would I need to flash one of those `*_APP_*` files to another location?

cjacker, 2024-11-29:
[Info] temp firmware-upgrade solution without windows.

Please refer to https://github.com/cjacker/opensource-toolchain-ch32v?tab=readme-ov-file#how-to-update-firmware-of-wch-linke

# for wch-link(ch549)
1, Enter ISP mode by short DP pin(P5.1) and 3v3 VCC pin when power up. 
2. use [ch552tool](https://github.com/MarsTechHAN/ch552tool) to erase and program the firmware:
```
ch55xtool -e -c
ch55xtool -f WCH-Link_APP_IAP_RV.bin
```

# for wch-linke(ch32v305fbp6)
1, prepare another WCH-LinkE and wire up the target WCH-LinkE as:
```
 +------------+            +-----------+
 |            |----3v3-----|           |
 |            |            |           |
 |            |----gnd-----|           |
 |            |            |  target   |
 | WCH-LinkE  |----dio-----| WCH-LinkE |
 |            |            |  to be    |
 |            |----clk-----|  update   |
 |            |            |           |
 |            |----nrst----|           |
 +------------+            +-----------+
```

2, hold IAP button of "target WCH-LinkE" down and plug WCH-LinkE to PC USB port. When power up, the "target WCH-LinkE" will enter IAP mode.

Official WCH-LinkE has a plastic case, you can not press the IAP button unless break the case. I make a [toggle IAP mode tool](https://github.com/cjacker/opensource-toolchain-ch32v/tree/main/firmwares/toggle_iap_mode), you can use it to toggle IAP mode of the target WCH-LinkE as:

> Plug the "target WCH-LinkE" to PC USB port first, And run 'switch_to_iap_mode'. 
> After the "target WCH-LinkE" was switched to IAP mode, unplug the target WCH-LinkE and plug WCH-LinkE to PC USB port, this time you don't have to hold the IAP button down of target WCH-LinkE.

3, update firmware:
```
wlink erase --method pin-rst --speed low --chip CH32V30X
wlink flash WCH-LinkE-APP-IAP.bin
```

cjacker, 2024-11-29:
Info: the process that WCH-LinkUtility update firmwares

After analysing McuCompilerDLL.dll:

1. OpenDevice: libusb_open/configuration/claim_interface. WCH-LinkUtility(windows) use ch375 functions.
2. setTargetChip: seems not used when update WCH-LinkE firmware, since it's target related.
3. read adapter type / version.
4. find corresponding chiptype and firmware, compare the version with 'wchlink.wcfg'.
5. If outdated, getIAPtype and setIAPmode.
    - malloc a buffer with size 0x80000 and load corresponding firmware to the buffer.
    - getIAPtype by write "0x02010f81" len 4 and read len 5, seems the result not used by WCH-LinkUtility.
    - setIAPmode by write "0x01010f81" len 4 and no read, since the "1a86:8010" disappear after setIAPmode.
7.  Sleep 1 and OpenDevice "4348:55e0", configuration and claim interface 0.
8. write "0x????0281" len 4 to endp 2, it seems erase-related action?  it's constructed by a 4 byte global static variable and 0x0281, but I didn't found where it initialise the global variable. 
9.  read len 2 from endp 2, 0x00 0x00 as expected and if not ch549.
10. allocate a buffer with size 64, memset to 0,  but everytime memcpy 60 from buffer. the last 4 keeps 0x00000000?
11. loop to write buf 64 to endp 2 everytime.
12. read 2, 0x00 0x00 as expected, I am not sure whether it should read everytime after write buf 64.
13. After written, write 0x????0283 to quit IAP mode, maybe 0x00000283? it's also constructed by a 4 byte global static variable and 0x0283, but I didn't found where it initialise the global variable.  and 0x00000283 really works.

And I didn't find how WCH-LinkUtility update ch549, it call McuCompiler_update(firmware_full_path, param_2 = 1), but in McuCompiler_update function, only handle the situation that param_2 = 0, maybe it's my tool's issue, some information lost after analysing.

cjacker, 2024-11-29:
Done. 

**wlink-iap**: A tool to upgrade / downgrade WCH-LinkE firmware online using IAP method.

please refer to: https://github.com/cjacker/wlink-iap

Hope wlink can implement this feature soon and deprecate my project.

andelf, 2024-11-30:
@cjacker Thanks. That's really helpful. Now I can rewrite the c code into Rust to impl this. 😆

cjacker, 2024-12-02:
> @cjacker Thanks. That's helpful. Now I can rewrite the c code into Rust to impl this. 😆

Please consider to add two options to involk enter iap mode and quit iap mode standalone . It seems still useful under some special situations:

1, to erase the whole data flash or/and reprogram the whole firmware with another wch-linke, it requires to enter IAP mode first by a physical button or by “ 0x01010f81” cmd. For official wchlinke, it has a case and not convenient to hold the IAP button down when power up.

2, user may accidently enter iap mode when using thirdparty wchlink since they all have no case, maybe user just want to switch rv/dap mode but hold the wrong button, I made such mistake before. And it has no chance to leave IAP mode anymore even with wch linkutility except reprogramming the firmware.

cjacker, 2024-12-04:
Update:

1, Before flash the firmware, 0x81020000 means erase the app code flash.
2, first loop is download loop, start with 0x80.
3, second loop is verify loop, start with 0x82.

21km43, 2025-01-17:
> > [@cjacker](https://github.com/cjacker) Thanks. That's helpful. Now I can rewrite the c code into Rust to impl this. 😆
> 
> Please consider to add two options to involk enter iap mode and quit iap mode standalone . It seems still useful under some special situations:
> 
> 1, To erase the whole data flash or/and reprogram the whole firmware with another wch-linke, it requires to enter IAP mode first by a physical button or by “ 0x01010f81” cmd. For official wchlinke, it has a case and not convenient to hold the IAP button down when power up. Some thirdparty device such as YD-Link even have no buttons on board.
> 
> 2, User may accidently enter iap mode when using thirdparty wchlink since they all have no case, maybe user just want to switch rv/dap mode but hold the wrong button, I made such mistake before. And it has no chance to leave IAP mode anymore even with wch linkutility except reprogramming the firmware.

I tried to implement functions which can enter or quit IAP mode and created PR.
https://github.com/ch32-rs/wlink/pull/84

Thanks for providing protocols of IAP mode.