ch32-rs/wlink #45: "chip doesn't support power off erase" error for ch32v003

thirstyone, 2023-12-26:
`wlink erase --chip ch32v003 --method power-off
00:06:01 [INFO] WCH-Link v2.11 (WCH-LinkE-CH32V305)
00:06:01 [INFO] Erase chip by PowerOff
00:06:01 [ERROR] --chip not specified
Error: Chip doesn't support power off erase`

Am I missing something?

andelf, 2023-12-26:
Could you provide full logs using wlink -vv?

thirstyone, 2023-12-26:
wlink -vv
15:23:18 [TRACE] (1) wlink::device: [src/device.rs:187] Acquired libusb context.
15:23:18 [INFO] Found WCH-LinkRV, Bus 250 Device 005: ID 1a86:8010
No command given, use --help for help.
hint: use `wlink status` to get started.

wlink -vv erase
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:48] Acquired libusb context.
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:79] Device: Bus 250 Device 005: ID 1a86:8010
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:83] Claimed interface 0 of USB device.
15:24:00 [DEBUG] (1) wlink::device: Serial number: "00A48F068577"
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:122] send command: GetProbeInfo
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810d01 01
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 820d04 020b1200
15:24:00 [INFO] WCH-Link v2.11 (WCH-LinkE-CH32V305)
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:122] send command: SetSpeed { riscvchip: 1, speed: High }
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810c02 0101
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 820c01 01
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:122] send command: AttachChip
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810d01 02
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 815501 01
15:24:00 [DEBUG] (1) wlink::operations: retrying...
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:122] send command: SetSpeed { riscvchip: 1, speed: High }
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810c02 0101
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 820c01 01
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:122] send command: AttachChip
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810d01 02
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 815501 01
15:24:00 [DEBUG] (1) wlink::operations: retrying...
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:122] send command: SetSpeed { riscvchip: 1, speed: High }
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810c02 0101
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 820c01 01
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:122] send command: AttachChip
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810d01 02
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 815501 01
15:24:00 [DEBUG] (1) wlink::operations: retrying...
Error: Probe is not attached to an MCU, or debug is not enabled. (hint: use wchisp to enable debug)

 wlink -vv erase --method power-off
15:25:15 [TRACE] (1) wlink::device: [src/device.rs:48] Acquired libusb context.
15:25:15 [TRACE] (1) wlink::device: [src/device.rs:79] Device: Bus 250 Device 005: ID 1a86:8010
15:25:15 [TRACE] (1) wlink::device: [src/device.rs:83] Claimed interface 0 of USB device.
15:25:15 [DEBUG] (1) wlink::device: Serial number: "00A48F068577"
15:25:15 [TRACE] (1) wlink::device: [src/device.rs:122] send command: GetProbeInfo
15:25:15 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810d01 01
15:25:15 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 820d04 020b1200
15:25:15 [INFO] WCH-Link v2.11 (WCH-LinkE-CH32V305)
15:25:15 [INFO] Erase chip by PowerOff
15:25:15 [ERROR] --chip not specified
Error: Chip doesn't support power off erase

wlink -vv erase --chip ch32v003 --method power-off
15:25:43 [TRACE] (1) wlink::device: [src/device.rs:48] Acquired libusb context.
15:25:43 [TRACE] (1) wlink::device: [src/device.rs:79] Device: Bus 250 Device 005: ID 1a86:8010
15:25:43 [TRACE] (1) wlink::device: [src/device.rs:83] Claimed interface 0 of USB device.
15:25:43 [DEBUG] (1) wlink::device: Serial number: "00A48F068577"
15:25:43 [TRACE] (1) wlink::device: [src/device.rs:122] send command: GetProbeInfo
15:25:43 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810d01 01
15:25:43 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 820d04 020b1200
15:25:43 [INFO] WCH-Link v2.11 (WCH-LinkE-CH32V305)
15:25:43 [INFO] Erase chip by PowerOff
15:25:43 [ERROR] --chip not specified
Error: Chip doesn't support power off erase

however: 
wlink -vv erase --chip "ch32v003j4m6" --method power-off
error: invalid value 'ch32v003j4m6' for '--chip <CHIP>': Unknown RISC-V Chip: 0x00

For more information, try '--help'.

andelf, 2023-12-27:
It seems the MCU is not connected.

> Probe is not attached to an MCU, or debug is not enabled.

The error message should be fixed.

thirstyone, 2023-12-27:
1. I checked continuity with a multimeter - everything is fine. 
2. I connected wchlinke to another chip from the same batch, which erased okay with default settings (i.e. without specifying --method), but when I tried to perform erase with power-off - I got the same error. 
3. Last time I had to connect it to a virtualbox windows, which erased it fine with power off through wchlink utility. Now I don't have virtualbox at hand. ;-(

andelf, 2023-12-27:
```
15:24:00 [TRACE] (1) wlink::device: [src/device.rs:122] send command: AttachChip
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:56] send 810d01 02
15:24:00 [TRACE] (1) wlink::transport: [src/transport.rs:47] recv 815501 01
```

The AttachChip command fails. At least `wlink status` or `wlink regs` should work.

thirstyone, 2023-12-27:
The problem is the target chip has disabled debug interface (hence I have to erase it via poweroff) - I corrupted it in the very beginning of my firmware code. I am not sure, status or regs would work in that case. 

Besides, is AttachChip needed when erasing via poweroff? (I don't see 810d01 command in that case in the log), and the error it returns is "Chip not specified" even when I specify it. Btw, when I use a 'wrong' partnumber, the error is different (see the log above)

thirstyone, 2024-01-05:
fiy: just connected that very chip to wchlink utility on a windowbox, and it erased on power off without any effort. Should I install wireshark?

thirstyone, 2024-01-05:
okay. here's what I get from the usb sniffer: 
81 0d 01 01 (sent to) 
82 0d 04 02 0b 12 00 (received from) 
81 0c 02 09 01 (sent to)
82 0c 01 01 (received from) 
81 0d 01 0f 09 (sent) 
82 0d 01 0f (received)

NB, there is no Attach Chip command before 81 0d 01 01

andelf, 2024-01-06:
Thanks. That's helpful.

andelf, 2024-01-07:
Could you check the new nightly release?

thirstyone, 2024-01-07:
so far so good, it works ;-)  however, the chip is not bricked: 
`wlink -vv erase --method power-off --chip ch32v003
10:28:29 [TRACE] (1) wlink::usb_device::libusb: [src/usb_device.rs:118] Device: Bus 250 Device 003: ID 1a86:8010
10:28:29 [DEBUG] (1) wlink::usb_device::libusb: Serial number: "00A48F068577"
10:28:29 [TRACE] (1) wlink::probe: [src/probe.rs:193] send command: GetProbeInfo
10:28:29 [TRACE] (1) wlink::probe: [src/probe.rs:174] send 810d01 01
10:28:29 [TRACE] (1) wlink::probe: [src/probe.rs:184] recv 820d04 020b1200
10:28:29 [INFO] Erase chip by PowerOff
10:28:29 [TRACE] (1) wlink::probe: [src/probe.rs:193] send command: SetSpeed { riscvchip: 9, speed: High }
10:28:29 [TRACE] (1) wlink::probe: [src/probe.rs:174] send 810c02 0901
10:28:29 [TRACE] (1) wlink::probe: [src/probe.rs:184] recv 820c01 01
10:28:29 [TRACE] (1) wlink::probe: [src/probe.rs:193] send command: ByPowerOff(CH32V003)
10:28:29 [TRACE] (1) wlink::probe: [src/probe.rs:174] send 810d02 0f09
10:28:29 [TRACE] (1) wlink::probe: [src/probe.rs:184] recv 820d01 0f`

okay. i just bricked it and successfully erased on power off! thank you!

closed 2024-01-07 by thirstyone