ch32-rs/wlink #46: Disable Flash Protection not working?

smaharj1, 2024-01-13:
I am pretty new to this, but connected to the CH32X035 through WCH-LinkE and tried the following command to unprotect the flash
```wlink unprotect```

I get the following message: 
```
wlink unprotect
22:51:20 [INFO] Connected to WCH-Link v2.11(v31) (WCH-LinkE-CH32V305)
22:51:20 [INFO] Attached chip: CH32X035 [CH32X035G8U6] (ChipID: 0x03560601)
22:51:20 [INFO] Unprotect Flash
22:51:20 [INFO] Flash protected: true
```

Am I missing something? My main goal is to try to disable Read Protection.

andelf, 2024-01-14:
This should be fixed. Could you check the newest changeset?
CH32X035 works. As I'm developing the https://github.com/ch32-rs/ch32x0-hal with this.

smaharj1, 2024-01-14:
Nice. I do not have access to the github link you posted, but getting the latest from the nightly release worked. Weird thing was both the nightly release version and standard cargo's wlink version are the same.

closed 2024-01-14 by smaharj1

andelf, 2024-01-15:
@smaharj1 nightly is triggered periodically or manually. 
I need to test on other MCUs, then do a cargo release.

After digging into the underlying flash protection mechanism, I discovered that four commands are used for protecting/unprotecting, with 2 commands each.