ch32-rs/wlink #47: WCH-Link underlying protocol

NgtNhat-131101, 2024-01-22:
Hi, I have an issue when I try using command `wlink status`. How can fix this issue?
Error: WCH-Link underlying protocol error: 0x55 [
    0x81,
    0x55,
    0x01,
    0x02,
]

andelf, 2024-02-19:
There's something wrong with the usb communication under Linux. 
I'll check this.

dominic-nhat-ng, 2024-03-21:
Thanks sir, I've solved that problem, and I have a question about CH32 flash usb, when I try flash program into my mcu, I can't understand why my usb sometimes update itself, and I need 2 device to update firmware for my version of mcu. I’m not change mcu, mcu series is CH32V003

andelf, 2026-11-12:
Closing as a duplicate of #94: the same `0x55` response means the probe could not connect to the target chip. Check the wiring and target power first; if those are fine, the debug interface is likely disabled (`wchisp config enable-debug`). #137 now adds that hint directly to the error message.

closed 2026-11-12 by andelf