ch32-rs/wlink #56: 似乎在连续写入两块不同的非零等等区域时会有问题？

kaidegit, 2024-03-14:
目前版本有写入到非零等待区域第二块时错误的问题，log如下。

```
wlink flash .\test.hex
08:08:17 [INFO] Connected to WCH-Link v2.10(v30) (WCH-LinkE-CH32V305)
08:08:17 [INFO] Attached chip: CH32V20X [CH32V203C8U6] (ChipID: 0x20300500)
08:08:17 [INFO] Chip UID: cd-ab-b7-5f-7a-bc-fe-c7
08:08:17 [INFO] Flash protected: false
08:08:17 [INFO] Read test.hex as IntelHex format
08:08:17 [INFO] Flashing 5240 bytes to 0x08000000
08:08:17 [INFO] Flash already unprotected
08:08:17 [INFO] Flash protected: false
██████████████████████████████████████████████████████████████████████████████████████████████████████████████ 5240/524008:08:17 [INFO] Flashing 54557 bytes to 0x08002000
08:08:17 [INFO] Flash already unprotected
08:08:17 [INFO] Flash protected: false
████████████████████████████████████████████████████████████████████████████████████████████████████████████ 54557/5455708:08:20 [INFO] Flashing 30944 bytes to 0x08010000
08:08:20 [INFO] Flash already unprotected
08:08:20 [INFO] Flash protected: false
████████████████████████████████████████████████████████████████████████████████████████████████████████████ 30944/3094408:08:22 [INFO] Flashing 35468 bytes to 0x0802e000
08:08:22 [INFO] Flash already unprotected
08:08:22 [INFO] Flash protected: false
Error: Error while fastprogram: [41, 01, 01, 05]
```

根据我的串口log可以看到烧写0x0802e000部分时固件已经启动了，跑了一点条之后变为了`Error: Error while fastprogram: [41, 01, 01, 05]`

我猜测可能是启动打断了烧录的问题，所以我尝试翻转了写入顺序

```rust
                        Firmware::Sections(sections) => {
                            // Flash section by section
                            if address != None {
                                log::warn!("--address is ignored when flashing ELF or ihex");
                            }
                            // reverse sections
                            let r_sections = sections.iter().rev().collect::<Vec<_>>();
                            for section in r_sections {
                                let start_address =
                                    sess.chip_family.fix_code_flash_start(section.address);
                                log::info!(
                                    "Flashing {} bytes to 0x{:08x}",
                                    section.data.len(),
                                    start_address
                                );
                                sess.write_flash(&section.data, start_address)?;
                            }
                        }
```

然后确实能正常烧写了，但是只成功烧录了最后一块，并且烧的还是错的，感觉问题主要在这一块。

```
.\wlink.exe flash test.hex
08:22:02 [INFO] Connected to WCH-Link v2.10(v30) (WCH-LinkE-CH32V305)
08:22:02 [INFO] Attached chip: CH32V20X [CH32V203C8U6] (ChipID: 0x20300500)
08:22:02 [INFO] Chip UID: cd-ab-b7-5f-7a-bc-fe-c7
08:22:02 [INFO] Flash protected: false
08:22:02 [INFO] Read test.hex as IntelHex format
08:22:02 [INFO] Flashing 435 bytes to 0x08037800
08:22:02 [INFO] Read protected: false
████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 435/43508:22:03 [INFO] Flashing 35468 bytes to 0x0802e000
08:22:03 [INFO] Read protected: false
████████████████████████████████████████████████████████████████████████████████████████████████████████████ 35468/3546808:22:05 [INFO] Flashing 30944 bytes to 0x08010000
08:22:05 [INFO] Read protected: false
████████████████████████████████████████████████████████████████████████████████████████████████████████████ 30944/3094408:22:06 [INFO] Flashing 54557 bytes to 0x08002000
08:22:07 [INFO] Read protected: false
████████████████████████████████████████████████████████████████████████████████████████████████████████████ 54557/5455708:22:09 [INFO] Flashing 5240 bytes to 0x08000000
08:22:10 [INFO] Read protected: false
██████████████████████████████████████████████████████████████████████████████████████████████████████████████ 5240/524008:22:10 [INFO] Flash done
08:22:10 [INFO] Now reset...

wlink dump 0x8000000 100
08:22:40 [INFO] Connected to WCH-Link v2.10(v30) (WCH-LinkE-CH32V305)
08:22:40 [INFO] Attached chip: CH32V20X [CH32V203C8U6] (ChipID: 0x20300500)
08:22:40 [INFO] Read memory from 0x08000000 to 0x08000064
Length: 100 (0x64) bytes | printable whitespace ascii_other non_ascii
08000000:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
08000010:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
08000020:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
08000030:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
08000040:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
08000050:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
08000060:   39 e3 39 e3                                          9×9×

wlink dump 0x0802e000 100
08:24:26 [INFO] Connected to WCH-Link v2.10(v30) (WCH-LinkE-CH32V305)
08:24:26 [INFO] Attached chip: CH32V20X [CH32V203C8U6] (ChipID: 0x20300500)
08:24:26 [INFO] Read memory from 0x0802e000 to 0x0802e064
Length: 100 (0x64) bytes | printable whitespace ascii_other non_ascii
0802e000:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
0802e010:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
0802e020:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
0802e030:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
0802e040:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
0802e050:   39 e3 39 e3  39 e3 39 e3  39 e3 39 e3  39 e3 39 e3   9×9×9×9×9×9×9×9×
0802e060:   39 e3 39 e3                                          9×9×

wlink dump 0x08037800 100
08:24:14 [INFO] Connected to WCH-Link v2.10(v30) (WCH-LinkE-CH32V305)
08:24:14 [INFO] Attached chip: CH32V20X [CH32V203C8U6] (ChipID: 0x20300500)
08:24:14 [INFO] Read memory from 0x08037800 to 0x08037864
Length: 100 (0x64) bytes | printable whitespace ascii_other non_ascii
08037800:   65 20 69 73  20 69 6e 76  61 6c 69 64  00 00 00 00   e is invalid0000
08037810:   73 68 6f 75  6c 64 20 6e  6f 74 20 72  65 61 63 68   should not reach
08037820:   20 68 65 72  65 00 00 00  4a 75 6d 70  5f 41 50 50    here000Jump_APP
08037830:   00 00 00 00  53 79 73 74  65 6d 49 6e  69 74 48 6f   0000SystemInitHo
08037840:   6f 6b 00 00  43 68 65 63  6b 41 70 70  56 61 6c 69   ok00CheckAppVali
08037850:   64 00 00 00  45 6e 74 72  79 41 50 50  00 00 00 00   d000EntryAPP0000
08037860:   00 01 02 02                                          0•••
```

[image]
[image]

kaidegit, 2024-03-15:
提供一个可以复现问题的hex

正常烧录的话log应该如下，串口PA9， 2M波特率

[image]

[test.zip](https://github.com/ch32-rs/wlink/files/14609004/test.zip)

andelf, 2024-03-19:
确认该问题.
需要确认下官方工具的协议时序.

commit b4d79532 (2024-03-30): fix: merge firmware sections, fix #56
closed 2024-03-30 by andelf