ch32-rs/wlink #57: Error: USB error: Entity not found

Censlab, 2024-03-21:
Hi, 

I use wlink with a WCH-Link dongle.

wlink alone give me : <WCH-Link#0 libusb device> Bus 002 Device 009 ID 1a86:8010(USB-FS 12 Mbps) (RV mode)

but  

wlink status give me : Error: USB error: Entity not found Caused by:  Entity not found

I work with Windows 11 23H2

What is this Entity ?
Thanks

kaidegit, 2024-03-22:
Try to use x86 version, or replace the WCH-Link (Interface 0)'s driver to winusb using zadig

Censlab, 2024-03-24:
Hi Kai,

This is what I did and indeed, with the Zadig driver, I immediately obtained the processor information. I think programming will work too. I will test in the next few days. THANKS

Date : 22/03/2024 05:51:29 Europe/Paris
Copie à : Censlab ***@***.***>;
   Author ***@***.***>

halfTree-dev, 2025-03-23:
I've met the same question on my Windows PC, and using x86 version works, Thanks.

NJCBY, 2025-06-10:
遇到同样问题，驱动换成Winusb后，官方的下载器就识别不到link了。

furrycoding, 2026-01-16:
Should probably put this WinUSB driver replacement step into the README

andelf, 2026-11-12:
This is a driver-binding problem: nusb can enumerate a WCH-Link bound to WCH's own driver but cannot open it, hence `Entity not found`.

wlink now supports WCH's native Windows driver (`WCHLinkDrv`, merged in #116) in addition to the legacy CH375 DLL and WinUSB, so rebinding the probe with Zadig is no longer required. The driver order and selector rules are documented in the README ("Windows drivers"), and #137 adds a troubleshooting entry.

Please try the next nightly/0.1.4 build. Reopen if `Entity not found` persists and include `wlink -vv list` plus the driver the probe is bound to.

closed 2026-11-12 by andelf