ch32-rs/wlink #60: wlink flash hangs when trying to reconnect serial console after reset

reinauer, 2024-03-30:
wlink  flash --watch-serial ./Build/app.elf  will hang and NOT reset the device.
wlink  flash  ./Build/app.elf will reset the device but of course not provide serial output.

I have a WCH Link and a WCH LinkE. Tried firmware v30 and v32. Same results.

DUT is a CH32V103C-R1-1v1.

On Windows with the WCH LinkTool I can reboot the DUT. But when using wlink the reset does not go through. The device itself hangs. I tried wlink on MacOS or Linux both. I can manually reset the device and it will come up and provide serial output through --watch-serial

If I don't specify --watch-serial the device will reboot fine.

```
$ sudo ~/bin/wlink  flash --watch-serial  ./Build/app.elf
17:27:30 [INFO] Connected to WCH-Link v2.12(v32) (WCH-LinkE-CH32V305)
17:27:30 [INFO] Attached chip: CH32V103 (ChipID: 0x2500410f)
17:27:30 [INFO] Chip ESIG: FlashSize(64KB) UID(cd-ab-76-0e-f4-bc-37-77)
17:27:30 [INFO] Flash protected: false
17:27:30 [INFO] Read ./Build/app.elf as ELF format
17:27:30 [INFO] Flashing 23268 bytes to 0x08000000
17:27:30 [INFO] Read protected: false
███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 23268/2326817:27:33 [INFO] Flash done
17:27:33 [INFO] Now reset...
```

```
17:26:15 [INFO] Flash done
17:26:15 [INFO] Now reset...
17:26:15 [TRACE] (1) wlink::probe: [src/probe.rs:196] send command: Soft
17:26:15 [TRACE] (1) wlink::probe: [src/probe.rs:177] send 810b01 01
17:26:15 [TRACE] (1) wlink::probe: [src/probe.rs:187] recv 820b01 01
17:26:15 [DEBUG] (1) wlink::probe: Opening serial port: "/dev/ttyACM0"
17:26:15 [TRACE] (1) wlink::probe: [src/probe.rs:271] Serial port opened: SerialPort ( name: /dev/ttyACM0 baud_rate: 115200 data_bits: Eight flow_control: None parity: None stop_bits: One )
```

andelf, 2024-03-31:
Does this happen every time?
I observed random fail on V203 too. I'm trying to resolve this.

reinauer, 2024-03-31:
Happens to me every time.