ch32-rs/wlink #65: 'wlink regs' resumes the CPU

OpusElectronics, 2024-05-24:
If the CPU is halted with:
```wlink halt```

then dumping registers with:
```wlink regs```

does resume the CPU. Not sure if this an intended behavior, but it bit me.

andelf, 2024-05-24:
A preferred way to fix this is to provide an in-app shell, holding the USB(probe) connection, making `halt`, `regs` a command.
When cmd exits it detaches.
No idea if `--no-detach` can fix this.

andelf, 2026-11-12:
This is a side effect of detaching: `detach_chip` resumes the MCU (which is why `halt` also disables detach). Pass the global `--no-detach` flag to keep the core halted:

```console
wlink halt
wlink --no-detach regs
```

#137 documents this in the README. Closing since the behaviour is intentional — reopen if `--no-detach` does not work for your setup.

closed 2026-11-12 by andelf