ch32-rs/wlink #68: "Invalid payload length" on WCH-Link CH549 at `ProbeInfo` response payload

rowanclarke, 2024-08-13:
Running `wlink -vv regs` gives the following output:

```
13:15:09 [TRACE] (1) wlink::usb_device::libusb: [src/usb_device.rs:118] Device: Bus 002 Device 003: ID 1a86:8010
13:15:09 [DEBUG] (1) wlink::usb_device::libusb: Serial number: "0001A0000001"
13:15:09 [TRACE] (1) wlink::probe: [src/probe.rs:196] send command: GetProbeInfo
13:15:09 [TRACE] (1) wlink::probe: [src/probe.rs:177] send 810d01 01
13:15:09 [TRACE] (1) wlink::probe: [src/probe.rs:187] recv 820d02 0201
Error: Invalid payload length
```

The error arises from the `ProbeInfo` response [here](https://github.com/ch32-rs/wlink/blob/fd31f2b614ab462a84d8508108445142c2af72ec/src/commands/control.rs#L29-L34), where there must be at least three bytes in the payload but my payload only contains two bytes [2, 1]. It then seems as though only the first two bytes are being used.

What should the payload of three bytes look like if only the first two are going to being used?

Thanks,
Rowan

fanoush, 2024-09-29:
Almost the same here
```
$ ./wlink -vv status
15:24:37 [TRACE] (1) wlink::usb_device::libusb: [src/usb_device.rs:118] Device: Bus 007 Device 031: ID 1a86:8010
15:24:37 [DEBUG] (1) wlink::usb_device::libusb: Serial number: "0001A0000001"
15:24:37 [TRACE] (1) wlink::probe: [src/probe.rs:196] send command: GetProbeInfo
15:24:37 [TRACE] (1) wlink::probe: [src/probe.rs:177] send 810d01 01
15:24:37 [TRACE] (1) wlink::probe: [src/probe.rs:187] recv 820d02 0203
Error: Invalid payload length
```
this is wch link with older combined firmware where you can still switch between  ARM CMSIS-DAP and RISCV mode by grounding TX pin when plugging device in.

When updating to newer RISCV only firmware it  works fine
```
$ ./wlink -vv status
15:31:09 [TRACE] (1) wlink::usb_device::libusb: [src/usb_device.rs:118] Device: Bus 007 Device 033: ID 1a86:8010
15:31:09 [DEBUG] (1) wlink::usb_device::libusb: Serial number: "434AC30F0659"
15:31:09 [TRACE] (1) wlink::probe: [src/probe.rs:196] send command: GetProbeInfo
15:31:09 [TRACE] (1) wlink::probe: [src/probe.rs:177] send 810d01 01
15:31:09 [TRACE] (1) wlink::probe: [src/probe.rs:187] recv 820d04 020c0100
15:31:09 [INFO] Connected to WCH-Link v2.12(v32) (WCH-Link-CH549)
15:31:09 [TRACE] (1) wlink::probe: [src/probe.rs:196] send command: SetSpeed { riscvchip: 1, speed: High }
15:31:09 [TRACE] (1) wlink::probe: [src/probe.rs:177] send 810c02 0101
15:31:09 [TRACE] (1) wlink::probe: [src/probe.rs:187] recv 820c01 01
15:31:09 [TRACE] (1) wlink::probe: [src/probe.rs:196] send command: AttachChip
15:31:09 [TRACE] (1) wlink::probe: [src/probe.rs:177] send 810d01 02
15:31:09 [TRACE] (1) wlink::probe: [src/probe.rs:187] recv 815501 01
15:31:09 [DEBUG] (1) wlink::operations: retrying...
...
```
the 0203 response is probably firmware version 2.03, newer one is 2.12 and the response is 4 bytes instead of 2

fourMew, 2024-09-30:
I encountered this same error using a CH549 based WCH-Link. Flashing a new firmware using the official WCH-LinkUtility and then switching to WinUSB drivers fixed it for me.

fanoush, 2024-09-30:
but anyway, the line
https://github.com/ch32-rs/wlink/blob/fd31f2b614ab462a84d8508108445142c2af72ec/src/commands/control.rs#L31
should probably have `<2` not `<3` so it is worth fixing if older firmware would otherwise work. As mentioned older firmware has both ARM and RISCV versions combined so if one uses devices with both cpu families and it otherwise works one may prefer to keep the older version.

ITstreet1, 2024-12-17:
> I encountered this same error using a CH549 based WCH-Link. Flashing a new firmware using the official WCH-LinkUtility and then switching to WinUSB drivers fixed it for me.

Where did you find the firmware for this one? I need it, too.

fanoush, 2024-12-18:
it is part of the wch-link utility zip from https://www.wch.cn/downloads/wch-linkutility_zip.html I think it is the WCH-Link_APP_IAP_RV.bin inside Firmware_Link folder if you want to flash risc-v chips and the WCH-Link_APP_IAP_ARM.bin if you want CMSIS-DAP functionality

ITstreet1, 2024-12-18:
> it is part of the wch-link utility zip from https://www.wch.cn/downloads/wch-linkutility_zip.html I think it is the WCH-Link_APP_IAP_RV.bin inside Firmware_Link folder if you want to flash risc-v chips and the WCH-Link_APP_IAP_ARM.bin if you want CMSIS-DAP functionality

Mate, thanks in a million!
I have this firmware from a year ago and didn't know what I have, LOL. This solved everything.