ch32-rs/wlink #69: Confusion between WCH Link vs WCH ISP and "modes"

dzervas, 2024-09-16:
Hey I just picked up a CH592 and CH582 boards from WeAct as well as the WCH-Link "mini.daplink". Links:

- [CH592F](https://www.aliexpress.com/item/1005006117859297.html?spm=a2g0o.order_detail.order_detail_item.3.4b77f19cAVFWVD)
- [CH582F](https://www.aliexpress.com/item/1005004784988010.html?spm=a2g0o.order_detail.order_detail_item.3.2c43f19chXp3rM)
- [WCH-Link (CH549)](https://www.aliexpress.com/item/1005003693318567.html?spm=a2g0o.order_detail.order_detail_item.5.4b77f19cAVFWVD)

My target is to blink a led with a rust.

I have quite a bit of experience with ARM and AVRs (back in the day) but I feel quite lost with all the different ways to debug/flash the MCU. My questions are:

1. Does the CH58x/59x support USB flashing? if so:
  - how? wch-isp?
  - do I need to flash a USB bootloader first?
  - what "type" of USB flashing does it support? DFU? Mass storage? something custom?
2. The wlink project targets the WCH-Link - so the debugger (like the SWD debuggers we have in ARM). What does wch-isp targets? USB or maybe something more specific like the ISP way to flash like the AVRs had (custom protocol)
3. How do I enable debug mode?
4. What does debug mode even mean? What if I turn it back off? Why would I do that? Does it lock the firmware?
5. I want to avoid running random binaries (WeAct studio, the moonriver IDE, etc.), is that even possible? Do I need them to enable debug mode?

qwertzdenek, 2024-09-17:
Hi,
I have some issues flashing CH592F from WeAct too. ISP tooling works as described is but WCH-Link running latest firmware is not able to communicate with MCU anyhow even with official tools (only serial console works). I executed the "Enable Simulation" process in ISPStudio but still MCU is not recognized. Is there anyone who was able to flash or control this MCU using WCH-Link?

I have official WCH-LinkE-R0-1v3 which should be compatible. Mentioned WCH-Link from WeAct does not support this board by the way.

dzervas, 2024-09-18:
Ok after some tinkering I've understood the following:

1. Yes, both support USB flashing and it's the main way to flash & configure the MCUs. It's a "closed" protocol implemented by wch-isp
2. WCH-ISP targets the custom USB protocol of the MCUs
3. The debug mode has to be enabled before wlink can be used. Do as follows:
   1. Hold the boot button
   2. Insert the MCU (NOT the wch-link) in the USB while holding the button
   3. Release the button
   4. `sudo ./target/release/wchisp config unprotect`
   5. Note that you have a small window (5s?) before the MCU goes back to normal mode. Have the command ready, insert the MCU and press enter.
4. It's essentially: debug port, read-out protection and maybe more? You can turn it back on but to turn it off the flash gets erased
5. Yeap, just use the wchisp to enable debug and then use wlink to debug the app

It all boils down to: USB is the first class citizen. It's the main way to flash and configure the MCU and not something that the bootloader manager. That's how you flash the firmware. The debug port is "only" for debugging and flashing without manual intervention. At least that's my understanding, I have a lot of more digging to do

closed 2024-09-18 by dzervas

biemster, 2025-03-06:
>    4. `sudo ./target/release/wchisp config unprotect`

This bricked my ch592, so use with care (https://github.com/ch32-rs/wchisp/issues/85). `config enable-debug` however worked as it should. After enabling this `wchisp` is not able to flash new firmware, I'm still waiting for my linkE to see if `wlink` is able in this mode. This is a very tricky mcu.

dzervas, 2025-03-10:
@biemster oh damn. I have no idea how/why but I added a warning to my comment

biemster, 2025-03-10:
> [@biemster](https://github.com/biemster) oh damn. I have no idea how/why but I added a warning to my comment

that's definitely a good idea I think. I actually managed to revive my board after I discovered that the bootloader is active for 10ms before resetting, it took some udev trickery to force a new config in that very tight time frame: https://gist.github.com/biemster/93f3fb932558063928562fd5ffb0cd1f

dzervas, 2025-03-10:
wow that's quite some work to bring it back to life - I secretly thought that "meh maybe they (you) didn't understand something/didn't try to re-enter debug/bootloader/whatever mode" but woah

now that I think of it I gotta try that on my seemingly fried 592F (I thought I fried it with 5V but maybe I don't remember right, it's been months)

biemster, 2025-03-10:
> now that I think of it I gotta try that on my seemingly fried 592F (I thought I fried it with 5V but maybe I don't remember right, it's been months)
> 

if `udevadm monitor -k -s usb` shows the bootloader coming alive every 0.5 seconds when connected this trick will very likely work.

cr1tbit, 2025-07-21:
I just wanted to leave some tips for the other people dealing with wchisp/wlink issues - in my case the target was ch582

* even after `unprotect` + `enable-debug` wch-link could not flash beyond 4k bytes. It seemed like the firmware on the probe was implemented incorrectly.
* flashing via wch-link**E** does work
* If the firmware seems not to run, try release builds

ntn888, 2025-08-16:
> Hi, I have some issues flashing CH592F from WeAct too. ISP tooling works as described is but WCH-Link running latest firmware is not able to communicate with MCU anyhow even with official tools [...]

I'm also having similar `Error: Probe is not attached to an MCU..` issue with WeAct ch592 board. Already enabled debug via wchisp. Though ch582 boards are working on the same setup.

I think the ch592 boards are faulty..