ch32-rs/wlink #83: uploader hardware

hvegh, 2025-01-16:
Do I need to buy the WCH-link hardware for this or can a generic swd ftdi adapter be used?
I have ftdi, stlink-v2 usb-blaster J-Link floating around here somewhere.

For now I just want to flash some code to an ch32v103

andelf, 2025-01-16:
You can use the USB-ISP tool: https://github.com/ch32-rs/wchisp .

WCH-Link is actually a CH32V305 dongle with close-source firmware.

hvegh, 2025-01-16:
Thanks for pointing me in the right direction.
I checked USB but it would not register, so I think need to set some specific boot configuration on the pins maybe...

I bought this one:
[image]

I wanted to experiment with this because chip can run on 5V which would be ideal for interfacing with some legacy stuff I think....

Sorry for misusing the issues section BTW.

hvegh, 2025-01-16:
Perfect!

```
wchisp info
17:55:10 [INFO] Opening USB device #0
17:55:10 [WARN] Find chip via alternative id: 0x33
17:55:10 [INFO] Chip: CH32V103(C8T6|C8U6|R8T6)[0x3f15] (Code Flash: 64KiB)
17:55:10 [INFO] Chip UID: xx-xx
17:55:10 [INFO] BTVER(bootloader ver): 02.70
17:55:10 [INFO] Code Flash protected: true
17:55:10 [INFO] Current config registers: ff00ff0000ff00ffffffffff00020700xx-xx
RDPR_USER: 0x00FF00FF
  [7:0]   RDPR 0xFF (0b11111111)
    `- Protected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
```

Love it let me know if I can help here...